makerbase-mks/MKS-Robin2 #2: Compile Error

frank-pian, 2020-05-14:
`fatal error: SoftwareSerial.h: No such file or directory`

makerbase-mks, 2020-05-21:
Hi! Use marlin2.0 firmware?

frank-pian, 2020-05-25:
@makerbase-mks yes

makerbase-mks, 2020-05-26:
Has your firmware enabled the driver type definition? Can you disable it and try to build it?

LordsWars, 2020-06-24:
Hello, I’ve been choosing a driver configuration for a week to compile your firmware :( it doesn’t work .... different errors all the time .... I changed in Configuration_adv - turning on the spindle and in pins_MKS_ROBIN2 I added pins for the spindle (pins from the E1 motor and the temperature table E1) .... the output of PF 2 - HEATER_1_PIN cannot be SPINDLE_LASER_PWM_PIN ????? in general redid the pins so that it would be possible to compile:
    #define SPINDLE_LASER_ENA_PIN PF2
    #define SPINDLE_LASER_PWM_PIN PA15
    #define SPINDLE_DIR_PIN PD6
   #define SPINDLE_STEP _PIN PD3
 and there’s a sea of ​​errors anyway .... is it possible to compile on your equipment and get a BIN file?
still prompt what versions of libraries you use ????
[env: mks_robin2]
platform = ststm32@5.3.0
framework = arduino ????????????
board = MKS_Robin2
extra_scripts = pre: buildroot / share / PlatformIO / scripts / mks_robin2.py
 buildroot / share / PlatformIO / scripts / mks_encrypt.py
build_flags = $ {common.build_flags}
  -DUSBCON -DUSBD_USE_CDC -DUSBD_VID = 0x0483 -DUSB_PRODUCT = \ "MKS_Robin2 \"
  -DTARGET_STM32F4 -DSTM32F407_5ZX -DVECT_TAB_OFFSET = 0xC000 -DARDUINO_ARCH_STM32 -std = gnu ++ 14
lib_deps = $ {common.lib_deps}
lib_ignore = Adafruit NeoPixel, c1921b4, TMC26XStepper, SailfishLCD, SailfishRGB_LED, SlowSoftI2CMaster
src_filter = $ {common.default_src_filter} + <src / HAL / HAL_STM32>
monitor_speed = 250000

LordsWars, 2020-06-24:
downloaded a clean version and tried to compile ...

Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/MKS_Robin2.html
PLATFORM: ST STM32 5.3.0 > STM32F407ZE (192k RAM. 512k Flash)
HARDWARE: STM32F407ZET6 168MHz, 128KB RAM, 502.23KB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, jlink, stlink)
PACKAGES:
 - framework-arduinoststm32 3.10500.190327 (1.5.0)
 - toolchain-gccarmnoneeabi 1.70201.0 (7.2.1)
Converting Marlin.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 12 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <U8glib-HAL> 0.4
|   |-- <Wire> 1.0
|   |-- <SPI> 1.0
|-- <LiquidCrystal> 1.3.4
|   |-- <Wire> 1.0
|-- <TMCStepper> 0.7.0
|   |-- <SPI> 1.0
|-- <30aa480>
|   |-- <Wire> 1.0
|-- <Arduino-L6470> 0.7.0
|-- <Servo> 1.1.2
|-- <SPI> 1.0
|-- <EEPROM> 2.0.1
|-- <IWatchdog> 1.0.0
|-- <Wire> 1.0
Building in release mode
Compiling .pioenvs\mks_robin2\src\src\SoftwareSerial.cpp.o
Marlin\src\SoftwareSerial.cpp:41:10: fatal error: avr/interrupt.h: No such file or directory
 #include <avr/interrupt.h>
          ^~~~~~~~~~~~~~~~~
compilation terminated.
Archiving .pioenvs\mks_robin2\libc1b\libU8glib-HAL_ID1932.a
Compiling .pioenvs\mks_robin2\lib682\TMCStepper_ID5513\source\CHOPCONF.cpp.o
Compiling .pioenvs\mks_robin2\lib682\TMCStepper_ID5513\source\COOLCONF.cpp.o
*** [.pioenvs\mks_robin2\src\src\SoftwareSerial.cpp.o] Error 1
In file included from .piolibdeps\mks_robin2\TMCStepper_ID5513\src\source\CHOPCONF.cpp:1:0:
.piolibdeps\mks_robin2\TMCStepper_ID5513\src/TMCStepper.h:25:11: fatal error: SoftwareSerial.h: No such file or directory

copied further received:

.piolibdeps\mks_robin2\TMCStepper_ID5513\src\SoftwareSerial.cpp:45:10: fatal error: util/delay_basic.h: No such file or directory
 #include <util/delay_basic.h>
          ^~~~~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pioenvs\mks_robin2\lib682\TMCStepper_ID5513\SoftwareSerial.cpp.o] Error 1

copied further received:

In file included from C:\Users\LORDS\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32/PinNames.h:5:0,
                 from C:\Users\LORDS\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32/interrupt.h:44,
                 from C:\Users\LORDS\.platformio\packages\framework-arduinoststm32\cores\arduino/board.h:7,
                 from C:\Users\LORDS\.platformio\packages\framework-arduinoststm32\cores\arduino/wiring.h:41,
                 from C:\Users\LORDS\.platformio\packages\framework-arduinoststm32\cores\arduino/Arduino.h:32,
                 from .piolibdeps\mks_robin2\TMCStepper_ID5513\src\SoftwareSerial.cpp:43:
.piolibdeps\mks_robin2\TMCStepper_ID5513\src\SoftwareSerial.cpp: In member function 'void SoftwareSerial::setTX(uint8_t)':
C:\Users\LORDS\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32/PortNames.h:79:45: error: invalid conversion from 'GPIO_TypeDef*' to 'uint8_t {aka unsigned char}' [-fpermissive]
 #define get_GPIO_Port(p) ((p < MAX_NB_PORT) ? GPIOPort[p] : (GPIO_TypeDef *)NULL)
C:\Users\LORDS\.platformio\packages\framework-arduinoststm32\cores\arduino/pins_arduino.h:232:38: note: in expansion of macro 'get_GPIO_Port'
 #define digitalPinToPort(p)         (get_GPIO_Port(STM_PORT(digitalPinToPinName(p))))
                                      ^~~~~~~~~~~~~
.piolibdeps\mks_robin2\TMCStepper_ID5513\src\SoftwareSerial.cpp:278:18: note: in expansion of macro 'digitalPinToPort'
   uint8_t port = digitalPinToPort(tx);
                  ^~~~~~~~~~~~~~~~
In file included from C:\Users\LORDS\.platformio\packages\framework-arduinoststm32\cores\arduino/Arduino.h:48:0,
[... 15 lines cut ...]
 #define get_GPIO_Port(p) ((p < MAX_NB_PORT) ? GPIOPort[p] : (GPIO_TypeDef *)NULL)
C:\Users\LORDS\.platformio\packages\framework-arduinoststm32\cores\arduino/pins_arduino.h:232:38: note: in expansion of macro 'get_GPIO_Port'
 #define digitalPinToPort(p)         (get_GPIO_Port(STM_PORT(digitalPinToPinName(p))))
                                      ^~~~~~~~~~~~~
.piolibdeps\mks_robin2\TMCStepper_ID5513\src\SoftwareSerial.cpp:289:18: note: in expansion of macro 'digitalPinToPort'
   uint8_t port = digitalPinToPort(rx);
                  ^~~~~~~~~~~~~~~~
In file included from C:\Users\LORDS\.platformio\packages\framework-arduinoststm32\cores\arduino/Arduino.h:48:0,
                 from .piolibdeps\mks_robin2\TMCStepper_ID5513\src\SoftwareSerial.cpp:43:
C:\Users\LORDS\.platformio\packages\framework-arduinoststm32\cores\arduino/pins_arduino.h:237:41: error: base operand of '->' is not a pointer
 #define portInputRegister(P)        (&(P->IDR))
                                         ^
.piolibdeps\mks_robin2\TMCStepper_ID5513\src\SoftwareSerial.cpp:290:26: note: in expansion of macro 'portInputRegister'
   _receivePortRegister = portInputRegister(port);
                          ^~~~~~~~~~~~~~~~~
.piolibdeps\mks_robin2\TMCStepper_ID5513\src\SoftwareSerial.cpp: In member function 'void SoftwareSerial::begin(long int)':
.piolibdeps\mks_robin2\TMCStepper_ID5513\src\SoftwareSerial.cpp:319:7: error: 'digitalPinToPCICR' was not declared in this scope
   if (digitalPinToPCICR(_receivePin)) {
       ^~~~~~~~~~~~~~~~~
.piolibdeps\mks_robin2\TMCStepper_ID5513\src\SoftwareSerial.cpp:319:7: note: suggested alternative: 'digitalPinToPort'
   if (digitalPinToPCICR(_receivePin)) {
       ^~~~~~~~~~~~~~~~~
       digitalPinToPort
compilation terminated due to -fmax-errors=5.
*** [.pioenvs\mks_robin2\lib682\TMCStepper_ID5513\SoftwareSerial.cpp.o] Error 1

LordsWars, 2020-06-24:
Dependency Graph
|-- <U8glib-HAL> 0.4
|   |-- <TMCStepper> 0.7.0
|   |   |-- <SPI> 1.0
|   |-- <Wire> 1.0
|   |-- <SPI> 1.0
|-- <LiquidCrystal> 1.3.4
|   |-- <TMCStepper> 0.7.0
|   |   |-- <SPI> 1.0
|   |-- <Wire> 1.0
|-- <TMCStepper> 0.7.0
|   |-- <SPI> 1.0
|-- <30aa480>
|   |-- <Wire> 1.0
|-- <Arduino-L6470> 0.7.0
|-- <Servo> 1.1.2
|-- <SPI> 1.0
|-- <EEPROM> 2.0.1
|-- <IWatchdog> 1.0.0
|-- <Wire> 1.0
Building in release mode