makerbase-mks/MKS-Robin2 #7: Robin2 :Layers out of alignment with PULSE_DELAY 5 #Pulse hold time (unit: machine cycle)

INNO-3D, 2020-09-08:
Hi,
I have experienced layer not aligned with PULSE_DELAY	5     #Pulse hold time (unit: machine cycle).
5 is anfortunatly the dafault value of Robin2_cfg.txt configuration file. 
I have not found any explanation about PULSE_DELAY parameter. I am currently using the value 50 with acceptable results
Can you explain what the correct value must be ?
[robin2_cfg.txt](https://github.com/makerbase-mks/MKS-Robin2/files/5189731/robin2_cfg.txt)

makerbase-mks, 2020-09-09:
Robin2 v1.0.3 has add it
[image]
[image]

INNO-3D, 2020-09-09:
I use firmware Robin2 v1.0.3  and #PULSE_DALAY 50 insted of 5. 
Why you have added #PULSE_DALAY parameter in the last version ?
What function does this parameter have?

makerbase-mks, 2020-09-10:
This parameter is mainly compatible with a variety of drivers. Some drivers, if this parameter is not set, the print will be crooked and misaligned.

INNO-3D, 2020-09-11:
How to set properly this parameter with TMC2209 ? Is there a guide ?