makerbase-mks/MKS-Robin2 #8: Problem Firmware Update

JanHH2019, 2020-09-21:
I have a mks robin2 v1.0 and wanted to install the latest firmware V1.0.3. Unfortunately that doesn't work, he loads the firmware, transfers the font and PIC files and then says "complete".
When I turn the board off and on again, the message booting comes up and nothing happens.
Have already tried the older versions of firmware V1.0.1 and V1.0.2, all with the same result.
What can I do now?

makerbase-mks, 2020-09-22:
Can you provide your Robin2 firmware updating process?

JanHH2019, 2020-09-22:
This version i have use first time ::

[https://github.com/makerbase-mks/MKS-Robin/blob/master/MKS%20Robin2/Mks_Robin2_V1.0.3%20Release%20file.rar](url)

I also tried these two versions ::

[https://github.com/makerbase-mks/MKS-Robin/blob/master/MKS%20Robin2/History%20version/Mks_Robin2_V1.0.2%20Release%20file.rar](url)

[https://github.com/makerbase-mks/MKS-Robin/blob/master/MKS%20Robin2/History%20version/Mks_Robin2_V1.0.2%20Release%20file.rar](url)

[image]
[image]
[image]
[image]
[image]
[image]
[image]

this is what the content of the sd card looks like after trying to boot ::
[image]

JanHH2019, 2020-09-22:
The problem is solved. I also wrote to Makerbase in the Facebook chat and got the tip there to format another SD card or to format the SD card. I have now done that and then the update runs through, it now also shows that it is reading in the PIC's and the fonts again and then the board starts correctly. Thank you for the effort..

closed 2020-09-22 by JanHH2019